JayHeng/NXP-MCUBootUtility #54: Read or Erase does sometimes not use correct parameters

ErichStyger, 2019-01-02:
I'm using the MCUBootUtility to erase and check the i.MXRT1064 memory (internal FLASH) at 0x7000'0000.
In some cases (not sure exactly when, sometimes it works, sometimes not) the address and size parameters are not respected correctly in the boot device memory page. 
https://mcuoneclipse.files.wordpress.com/2019/01/wrong-read-memory.png?w=584&h=370
The usual sequence is that I'm simply connecting to the device and try to read or erase that memory.

Let me know if you need any more information.

JayHeng, 2019-01-03:
Hi, the root cause is that sometimes tool doesn't get correct memory info from device, you can see wrong page/sector/memory size ((all 0xffffffffs)) in device status window (lower left quarter)

JayHeng, 2019-01-03:
I guess there is cache issue in ROM flashloader, tool reads device memory(FDCB) by ROM flashloader to get memory info, it seems that sometimes flashloader returns all 0xFFs data.

ErichStyger, 2019-01-03:
Ah, I see. And yes, I can confirm that the issue indeed is present if the device memory FDCB returns all 0xff. I can reproduce this with if I erase the FLASH.

closed 2020-06-16 by JayHeng