JayHeng/NXP-MCUBootUtility #56: Fails to generate keys and certs in "HAB signed image boot" mode on Windows

ricynlee, 2019-01-06:
- After the `Generate Certificate, SRK` button is pressed, the "tools\cst\3.0.1\keys" **FILE** is created instead of a folder. So nothing is copied to the directory.
> Manually replace the mentioned file with a folder, so "key_pass.txt" and "serial" can be copied to the folder. However, the "tools\cst\3.0.1\crts" directory won't be created. Manual creation won't help generate certificates.

JayHeng, 2019-01-06:
The formal v1.0.0 of NXP-MCUBootUtility doesn't contain CST tool due to the restriction of export control on security product, please follow this blog ( https://www.cnblogs.com/henjay724/p/10189593.html ) to add CST into NXP-MCUBootUtility, or HAB related function will not be enabled.

JayHeng, 2019-01-06:
Tell me your email, so I can help you further

closed 2019-01-07 by JayHeng