JayHeng/NXP-MCUBootUtility #57: Fail to generate bootable image from MCUXpress .axf file

ripe909, 2019-01-06:
I have tried multiple projects (blinky) and output formats (axf and hex) but always the NXPMCUBootUtility shows an error that "Bootable image is not generated successfully!"

Is there something special that should be done to use .axf output?

JayHeng, 2019-01-07:
NXP-MCUBootUtility supports .axf file from MCUX, you can try \NXP-MCUBootUtility\apps\NXP_MIMXRT1060-EVK_Rev.A1\led_blinky_0x0000a000_mcux.axf on your side, note that you have to select '.out(axf) from MCUXpresso' in image format selection initiatively.

ripe909, 2019-01-07:
Sure, here is my output .axf file:
[evkbimxrt1050_led_blinky_mcuboot.axf.zip](https://github.com/JayHeng/NXP-MCUBootUtility/files/2733154/evkbimxrt1050_led_blinky_mcuboot.axf.zip)
I am using MCUXpresso 10.2 with SDK 2.4.1

JayHeng, 2019-01-07:
I saw that there were i.MX headers(FDCB, IVT...) in your .axf file. In principle, you should provide bare image (without any headers) for NXP-MCUBootUtility, tool can add necessary headers automatically.

You can set XIP_BOOT_HEADER_ENABLE to 0 in porject option then rebuild the project, see detail in User Guide “2 Preparation”

ripe909, 2019-01-07:
Thanks!  That was it...  I didn't notice the XIP directive had to be changed.  I modified and tried again successfully.

closed 2019-01-07 by ripe909