JayHeng/NXP-MCUBootUtility #58: Support for Arduino (Teensy 4) built binaries?

ripe909, 2019-01-07:
Teensy 4 is based on the IMXRT1050/60 and they have started beta development of the toolchain.  I'm not sure how feasible or necessary it is, but it would be interesting to support the hex/elf output files from the Arduino build output.

JayHeng, 2019-01-08:
Is Teensy4 board name or toolchain name? NXP-MCUBootUtility is board-independent, it should support all kinds of i.MXRT boards, because tool only talk with the onchip ROM of i.MXRT. and for binary format, hex/srec/bin are common formats, they are toolchain-independent.

The only issue is elf file from dedicate toolchain, if it is not common GNU ELF format, NXP-MCUBootUtility may not support it.

ripe909, 2019-01-08:
Arduino/Teensy4 uses GCC, I am attaching a Blinky example .elf file built from Arduino using the Teensy 4 core.  I tried reading it in MCUBootUtility unsuccessfully.
[Blink.pde.elf.zip](https://github.com/JayHeng/NXP-MCUBootUtility/files/2736885/Blink.pde.elf.zip)

JayHeng, 2019-01-08:
the same root cause as your last question, this elf file contains i.MXRT headers

ripe909, 2019-01-08:
cool, I may wait for your future updates then.  In general the Arduino build can't/shouldn't be changed.

I also saw your blog postings (via Google translate)  they are very good and helpful!

JayHeng, 2019-01-08:
so it seems to be a very common issue, v1.1 will be released by end of this week

closed 2019-01-25 by JayHeng