JayHeng/NXP-MCUBootUtility #59: End-user mode.

ripe909, 2019-01-10:
I would love to see a version/mode for end users performing field updates.  

This would be a very stripped down user interface which only allows selecting a firmware file and then initiating the all-in-one action.  All of the options and config would be hidden and preselected (based on JSON config file?) for the specific device they are updating.   

I started to look at doing this myself, but it may be something that others could benefit from as well.

[image]

JayHeng, 2019-03-17:
What is your NXPISP status? I will start to create this kind of version (called RT-Programmer) in next week

ripe909, 2019-03-18:
Great news!   

For ISP I have a selection, J-Link, LPC-Link, and NXP-EVK OpenSDA, as well as a Beagle USB 480 for USB debugging if that helps.

JayHeng, 2019-04-18:
The end-user mode version is done, v1.0.0 has been released
https://github.com/JayHeng/RT-Flash

closed 2019-04-18 by JayHeng