JayHeng/NXP-MCUBootUtility #65: Document how to replace flashloader with custom compiled version

ripe909, 2019-03-27:
If I wish to customize the flashloader used (for IMXRT1050), how can this be changed out in NXP-MCUBootUtility?

I tried generating a new flashloader.elf and flashloader.srec and placing them in ./src/targets/MIMXRT1052  but it looks like these are not used, instead the ivt_flashloader.bin file is used (?)

I am trying to generate the new custom flashloader in MCUXpresso IDE using the SDK example.

JayHeng, 2019-03-29:
yes, you need to provide new ivt_flashloader.bin, it is used for DEV Unsigned Image boot

JayHeng, 2019-03-29:
flashloader.srec is used for Signed Image Boot, because tool will generate signed flashloader based on flashloader.srec

ripe909, 2019-03-29:
Thanks, I will give it a shot and see if I am successful.

ripe909, 2019-08-20:
I was able to generate a new flashloader and use it with NXP_MCUBootUtility.  I documented it here: [https://community.nxp.com/thread/500209](url)

closed 2020-10-12 by JayHeng