JayHeng/NXP-MCUBootUtility #103: Behavior when QE bit is not configured.

NorihiroMichigami, 2019-06-19:
Hello, 

It's first time to post an issue to Github, 
So please correct me if my issue report should not be discussed here.

We use following two QSPI flash memory with MCUBootUtility.
We _could_ successfully flash these memory, but behavior of MCUBootUtility
when QE bit is NOT CONFIGURED differs between these two parts.

 Winbond:    W25Q80DVUXIE
 GigaDevice: GD25Q40CEIGR

When we don't configure QE bit from your tool for GD25Q40CEIGR, we can't flash it.
This is expected result because we don't specify QE bit in your tool. 
If we specify QE bit correctly, we can flash GD25Q40CEIGR via QSPI mode, of course. 

When we don't configure QE bit from your tool for W25Q80DVUXIE, 
we can flash it via QSPI bus correctly, even QE bit is not specified in your tool. 
Our expected result is that we can't flash this memory as GD25Q40CEIGR case.
If we specify QE bit in your tool, we can flash W25Q80DVUXIE via QSPI bus as well. 

Note 1)
Before flashing those memory, we confirmed QE bit in those flash memory was "0" with debugger in advance.
So we are sure that those memory operate in single SPI mode when our system boot up
 
Note 2)
Before flashing those memory, we removed your tool from our PC 
completely and setup it again. 
We did this so that we could make sure that there was no valid .json file 
which initialized your tool when we invoked. 

I understand that this may not be critical issue.
But we plan to use this nice tool at our factory to program our firmware in QSPI flash,
So we want your nice tool to work consistently for any flash memory, any flash memory vendors.  

Thanks,
Norihiro Michigami

JayHeng, 2019-06-19:
Hi,

  Thanks for your feedback.
  It is about SFDP version in Flash. if the SFDP version is JESD216A/B, then Flashloader can detect QE bit status, if it is not set, Flashloader will set it automatically. if SFDP version is not JESD216A/B, then user need to specify QE bit location in my tool manually.

Best Regards,
Jay

NorihiroMichigami, 2019-06-19:
Hello Jay-san, 

Thank you for your reply. 
SFDP is likely cause of different observation we see.

I quickly checked the datasheet for both Winbond and GigaDevice.
Both parts seem to support SFDP, but I can't find SFDP version number from datasheet.
Could you advise me which your source code checks SFDP version number?

Thanks,
Norihiro Michigami
AVNET

JayHeng, 2019-06-19:
The SFDP version is in SFDP header, you can find detailed format of SFDP header in JESD216 spec.
Tool gets SFDP header from Flash by issuing Read SFDP Register command.

JayHeng, 2019-06-19:
'SFDP Minor Revision' member in SFDP header shows SFDP version

    kSfdp_Version_Minor_0 = 0, // JESD216
    kSfdp_Version_Minor_A = 5, // JESD216A
    kSfdp_Version_Minor_B = 6, // JESD216B
    kSfdp_Version_Minor_C = 7, // JESD216C

NorihiroMichigami, 2019-06-20:
Hello Jay, 

Thank you for your information.
I can't find SFDP minor version in Windbond's data sheet. 
But I found that SFDP minor version of GigaDevice was "0x0" in their data sheet.
So at least for GigaDevice, your answer explains our observation on this device.

Thanks,
Norihiro Michigami
AVNET

NorihiroMichigami, 2019-07-02:
Hello Jay,

Eventually, we could confirm that Winbond’s flash complied with JESD216A. 
So your code perfectly worked as you designed. 

Thanks,
Nori

closed 2019-07-02 by NorihiroMichigami