JayHeng/NXP-MCUBootUtility #110: 无法打开，点击mcubootutility直接关了

lyw851230, 2019-07-02:
双击后，直接关闭，显示ImportError:cannot import name RTyyyy_uidef
                                      [6356]Failed to execute script main

lyw851230, 2019-07-02:
双击后，直接关闭，显示ImportError:cannot import name RTyyyy_uidef
[6356]Failed to execute script main

JayHeng, 2019-07-02:
需要下载release版 https://github.com/JayHeng/NXP-MCUBootUtility/releases

JayHeng, 2019-07-02:
master一直处于开发状态，bin文件夹下面的exe没有实时编译

lyw851230, 2019-07-03:
下载了release版本，正常打开，连接上USB线后，弹出VID=0x1Fc9,PID=0x0130。设置我的nor flash为IS25Lp064A，点击connect，弹出cannot find USB-HID device (vid=0x15A2,pid=0x0073),please connect USB cable to your board first!这是怎么回事啊，我的USB线连接着呢。说明下，我是keil开发的，

lyw851230, 2019-07-03:
您好：
       我下载了release版本后，按照https://mp.weixin.qq.com/s/D2UxyYNpUNdX2mMFBV3m7g上的说明进行了设置，并且连接成功（指示灯显示蓝色）。然后点“一键启动”，再用scan和view bootable image，如下两图所示：view bootable image未看到任何信息，转到internal boot模式后没有看到程序启动。请帮忙，谢谢。
刘亚伟  13821665625

lyw851230@163.com
 
发件人： 衡杰
发送时间： 2019-07-02 16:39
收件人： JayHeng/NXP-MCUBootUtility
抄送： lyw851230; Author
主题： Re: [JayHeng/NXP-MCUBootUtility] 无法打开，点击mcubootutility直接关了 (#110)
master一直处于开发状态，bin文件夹下面的exe没有实时编译

closed 2019-07-03 by JayHeng