JayHeng/NXP-MCUBootUtility #111: cannot find USB-HID device (vid=0x15A2,pid=0x0073),

lyw851230, 2019-07-03:
下载了release版本，正常打开，连接上USB线后，弹出VID=0x1Fc9,PID=0x0130。设置我的nor flash为IS25Lp064A，点击connect，弹出cannot find USB-HID device (vid=0x15A2,pid=0x0073),please connect USB cable to your board first!这是怎么回事啊，我的USB线连接着呢。说明下，我是keil开发的，

JayHeng, 2019-07-03:
您好，这是没能跳转到Flashloader里。您用的具体哪个RT芯片型号？FlexRAM有没有重新配置过？

lyw851230, 2019-07-03:
您好，我用的型号是：MIMXRT1052CVL5B/BGA-196，FlexRAM没有重新配置过。一个月之前按照这篇文章
https://mp.weixin.qq.com/s/elCQyYIZOJcHv7er2yvX_Q    

尝试过blhost和MFGTools连接过。当时还再NXP论坛里发了这篇帖子，但是问题没有解决。

http://www.nxpic.org/module/forum/thread-616626-1-2.html 

请帮忙分析一下。

刘亚伟   2019年7月3号夜   13821665625

lyw851230@163.com
 
From: 衡杰
Date: 2019-07-03 21:41
To: JayHeng/NXP-MCUBootUtility
CC: lyw851230; Author
Subject: Re: [JayHeng/NXP-MCUBootUtility] cannot find USB-HID device (vid=0x15A2,pid=0x0073), (#111)
您好，这是没能跳转到Flashloader里。您用的具体哪个RT芯片型号？FlexRAM有没有重新配置过？

JayHeng, 2019-07-04:
目前官方发布的RT105x Flashloader是v1.1版本，其有UART接口干扰缺陷（如果没有上拉的话），会影响USB的枚举。目前\NXP-MCUBootUtility-master\src\targets\MIMXRT1052\ivt_flashloader.bin文件是修复了这个问题的，你试一下

JayHeng, 2019-07-04:
MCUBootUtility v1.4.0和RT-Flash v2.0.0都是用的官方RT105x Flashloader v1.1版本，所以会存在同样的问题。计划在MCUBootUtility v2.0.0和RT-Flash v2.1.0里正式修复这个问题

JayHeng, 2019-07-04:
你可以加到 恩智浦MCUBootUtility工具交流qq群里（群号：867386086），沟通可以更即时一些

lyw851230, 2019-07-04:
针对您的回复，我有两个问题：
1.若是我将串口1端口TX(GPIO_AD_B0_12)、RX(GPIO_AD_B0_13引脚)上拉，USB的枚举将不会有问题？
2.串口上拉会屏蔽串口为下载口？这个是什么机理？

lyw851230@163.com
 
From: 衡杰
Date: 2019-07-04 13:40
To: JayHeng/NXP-MCUBootUtility
CC: lyw851230; Author
Subject: Re: [JayHeng/NXP-MCUBootUtility] EXT-13 Cannot find USB-HID device (vid=0x15A2,pid=0x0073) (#111)
目前官方发布的RT105x Flashloader是v1.1版本，其有UART接口干扰缺陷（如果没有上拉的话），会影响USB的枚举。目前\NXP-MCUBootUtility-master\src\targets\MIMXRT1052\ivt_flashloader.bin文件是修复了这个问题的，你试一下

JayHeng, 2019-07-04:
1.是的，加上拉就不会有问题
2.Flashloader会轮询UART和USB，如果UART先生效，将禁掉USB

closed 2019-07-05 by JayHeng