JayHeng/NXP-MCUBootUtility #124: Replicating MCUBootUtility actions manually with NXP commandline tools

ripe909, 2019-08-20:
Jay,

I am trying to manually perform all of the steps that NXP-MCUBootUtility is executing using the commandline only (without NXP-MCUBootUtility). 
I am using sdphost, blhost, objcopy and elftosb utilities.

I am generating an .axf file in MCUXpresso 11.0.0 and then using objcopy and elftosb utility to create the .bin file  using the same commands and .bd that NXP-MCUBootUtility is using.

`./elftosb -f imx -V -c /Volumes/Untitled/imx_application_gen.bd -o ivt__led_blinky_unsigned.bin evkbimxrt1050_iled_blinky_mcuboot.srec`

Is NXP-MCUBootUtility doing something else to the  .bin file after generating it with elftosb?
Looking at the log, I think I am missing the step that uses the flexspiNorCfg.dat file being read from the device flash...   Is this in place of using dcd.bin with the .bd file when generating the .bin file with elftosb?

If so, how is NXP-MCUBootUtility using flexspiNorCfg.dat to modify the .bin file?

ripe909, 2019-08-21:
I continue to try and track down differences in my manual process.  

It seems that the .srec file being generated on my Mac is a bit different (on the first line) than the one generated by NXP-MCUBootUtility on windows.  This is using the same .axf file to generate the .srec file.

I am still not sure how/why the flash is being read into flexspiNorCfg.dat

ripe909, 2019-08-21:
I think the Mac compiled version of elftosb has some bugs.   If I use the Windows version of elftosb from the command line, the .bin output works as expected...

Where did the Mac version of elftosb in the NXP-MCUBootUtility come from?  Is it tested to actually work?

AshUK, 2019-10-04:
@ripe909 did you manage write a script to program your nxp chip in the end? im trying to do the same, i want to program it via command line.

ripe909, 2019-10-13:
> 
> 
> @ripe909 did you manage write a script to program your nxp chip in the end? im trying to do the same, i want to program it via command line.

Yes, I was able to get it working, not with a script, but using XCode on a Mac...  although elftosb doesn't work on the Mac for IMXRT yet.

JayHeng, 2019-11-25:
Please report the mac version of elftosb issue to NXP community

closed 2019-11-25 by JayHeng