JayHeng/NXP-MCUBootUtility #132: Cannot run MCUBoot under Win 7 64 bit

nospamsam2011, 2019-10-04:
Sadly, there appears to be an error running MCUBoot under Win 7 64-bit.  The error received is as follows from the console:

> C:\Software\NXP\NXP-MCUBootUtility-master\bin>nxp-mcubootutility.exe
Traceback (most recent call last):
  File "main.py", line 467, in <module>
  File "main.py", line 50, in __init__
  File "_main\RTxxx_main.py", line 21, in __init__
  File "mem\RTxxx_memcore.py", line 20, in __init__
  File "run\RTxxx_runcore.py", line 59, in __init__
  File "gen\RTxxx_gencore.py", line 25, in __init__
  File "ui\RTxxx_uicore.py", line 18, in __init__
  File "_main\RTyyyy_main.py", line 38, in __init__
  File "mem\RTyyyy_memcore.py", line 20, in __init__
  File "fuse\RTyyyy_fusecore.py", line 19, in __init__
  File "run\RTyyyy_runcore.py", line 73, in __init__
  File "gen\RTyyyy_gencore.py", line 24, in __init__
  File "ui\RTyyyy_uicore.py", line 22, in __init__
  File "mem\memcore.py", line 20, in __init__
  File "run\runcore.py", line 17, in __init__
  File "gen\gencore.py", line 17, in __init__
  File "ui\uicore.py", line 95, in __init__
  File "ui\uicore.py", line 267, in _initPortSetupValue
  File "run\RTyyyy_runcore.py", line 110, in RTyyyy_getUsbid
  File "run\RTyyyy_runcore.py", line 107, in RTyyyy_createMcuTarget
  File "run\RTyyyy_runcore.py", line 61, in RTyyyy_createTarget
  File "C:\Software\NXP\NXP-MCUBootUtility-master\src\targets\MIMXRT1052\bltarge
tconfig.py", line 73, in <module>
    registerAddrDict = RTyyyy_rundef.registerAddrDict_RT10yy
AttributeError: 'module' object has no attribute 'registerAddrDict_RT10yy'
[10180] Failed to execute script 

As mentioned in the notes, I installed Python 2.7.15 32bit and executed the env scripts, and I get the same error when I rerun MCUBoot.  Does anyone know what I am doing wrong?

Thanks in advance!

Cheers,

Sam

nospamsam2011, 2019-10-05:
Good day All,
As a follow up, I tried this under Windows 10 -64 bit and I am getting the same error.
Cheers,
Sam

nospamsam2011, 2019-10-06:
Good day All,
I found the issue to this... I have a newer version of Python also installed (3.6.x) and it would appear that it was being called for the application.  I tried a few things to remedy this, but to no avail.  Ultimately I installed a virtual Python2.7 environment (e.g. virtualenv), activated it, ran the scripts under the env directory, and then command line executed the resulting new app and I now have the application running... So, all is good.
Cheers,
Sam

closed 2019-10-06 by nospamsam2011

ben-github, 2019-10-30:
I did a little digging into this issue as it is weird. RTyyyy_rundef has all the attributes from the RTyyy_rundef.py file **except** the dictionaries. It gets the contents of those dictionaries (e.g. kRegisterMask_SRC_SBMR2_Bmod) directly in the module which is also strange, but it does not get the dictionaries themselves. I cannot explain this behavior and isn't a python3 thing. I'm guessing it is related to the python packaging / pyinstalled since I cannot recreate it with python3 when I run the program directly with python main.py. 

For those interested in running this on python3, I made some changes to support python3 -- the pull request is at https://github.com/JayHeng/NXP-MCUBootUtility/pull/136 and the code at https://github.com/ben-github/NXP-MCUBootUtility/tree/Python3 (Python3 branch). I haven't tested it with the binary, just by running 'python main.py' so you'll need to install wxpython and some other dependencies.