JayHeng/NXP-MCUBootUtility #133: Secondary Pinmux is not working with RT1021 device

nospamsam2011, 2019-10-08:
Good day All,
I have been trying to program the serial Flash connected on the RT1021's secondary pinmux, but MCUBootUtility does not appear to be communicating with the serial flash device.  The serial Flash device is the same as used on the RT1021's EVK and I have configured MCUBootUtility's "Boot Device Configuration" with "Has Option2:  Yes", "Flash Connection: Single Port A", and "Enable Second Pinmux: Yes".  Using a logic analyzer when I try click the "reconnect" button there is no signal activity to the flash device.  The device status window indicates that MCUBootUtility is communicating with the RT1021, as I can see that it reads the following:

--------MCU device Register----------
OCOTP->UUID[31:00] = 0x65f82921
OCOTP->UUID[63:32] = 0x2f5271d2
SRC->SBMR1 = 0xe
SRC->SBMR2 = 0x1000001
BMOD[1:0] = 2'b01 (Serial Downloader)
HAB status = Open
--------MCU device eFusemap--------
(0x450) BOOT_CFG0 = 0x0
(0x460) BOOT_CFG1 = 0x0
(0x470) BOOT_CFG2 = 0x0
BT_FUSE_SEL = 1'b0
  When BMOD[1:0] = 2'b00 (Boot From Fuses), It means there is no application in boot device, MCU will enter serial downloader mode directly
  When BMOD[1:0] = 2'b10 (Internal Boot), It means MCU will boot application according to both BOOT_CFGx pins and Fuse BOOT_CFGx
----------FlexRAM memory-----------
IOMUXC_GPR->GPR16 = --------
`

I tried both the latest version of MCUBootUtility (2.1.0) and the past version (2.0.0) and I get the same results under Windows 7-64 bit.

If I place boot mode to "internal boot" I can see activity to the serial Flash device and so I know my logic analyzer is working.

Has anyone else had this issue?

Thanks in advance!

Cheers,

Sam

nospamsam2011, 2019-10-25:
Good day All,
I have recently been working with NXP and there indeed appears to be a bug with regards to MCUBootUtility programming the flash on the secondary pinmux.  As soon as I know more and/or have a fix I will post back.
Cheers,
Sam

JayHeng, 2019-11-25:
There is a bug about second pinmux in RT1020 Flashloader, it is not tool issue