JayHeng/NXP-MCUBootUtility #145: BEE加密配置模块不显示

gyboy, 2019-12-02:
从release部分下载下来的2.1正式版本，配置完HAB和BEE之后打开工具不显示BEE的配置项，似乎是要激活一下，附上生成的默认配置文件。

[nsb_settings.txt](https://github.com/JayHeng/NXP-MCUBootUtility/files/3909070/nsb_settings.txt)

[image]

JayHeng, 2019-12-02:
这是个已知的UI界面显示问题，我的电脑上能正常显示的。

JayHeng, 2019-12-02:
[image]

gyboy, 2019-12-02:
> 
> 
> 这是我电脑上（Latitude E7470, Win10, x64）的显示
> [image]
旧的版本在我电脑上可以显示，但是最近的几个版本就会出现这样的问题，我在别的PC和笔记本上都试了，都是一样的问题，有修复的计划么，不然新版本可能无法使用，谢谢。

closed 2019-12-12 by JayHeng